verlab/hero_common #11: roslaunch hero_gazebo gazebo_wizard.launch  : Launches empty Gazebo with no UI wizard

sgtmahoy, 2022-05-03:
Hey all, great stuff so far! Wondering if you have any insight to the subject issue. If possible when creating your how-to, would it be possible to include a brief walkthrough from a github pull to a gazebo sim? 

roslaunch hero_gazebo env_spawn.launch spawns the arena and hero bots if gazebo is already launched, however none of the scripts link to the simulated swarm when run.

Thanks!

rezeck, 2022-05-03:
Hey @sgtmahoy 
Thanks for your interest in this repo!

Sorry for the lack of instructions so far. I'm trying to constantly add some instructions when I have some time.
I'll try to upload a walkthrough tutorial from a github pull to a gazebo as soon as possible.

If I understand correctly, you are having a problem with the UI wizard, right?
Did you note any warnings or errors in your terminal? 
Try running the script,
```sh
rosrun hero_gazebo hero_wizard.py
```
and check for any errors. Also, check if the hero_wizard.py file has executable permissions.
```sh
chmod +x hero_wizard.py
```

I haven't added the package's dependencies yet, and this UI was created using QT, so make sure the python3-pyqt5 package is installed. Use python3-pyqt5 for ROS noetic, and python-pyqt5 for ROS melodic.
You can use apt or similar to install:

```sh
sudo apt-get install python3-pyqt5
```

To advance your tests, here is a short video showing step-by-step guide for first use of hero wizard:
<div align="center">
      <a href="https://www.youtube.com/watch?v=IO7FPEwkZkE">
     <img 
      src="https://img.youtube.com/vi/IO7FPEwkZkE/1.jpg" 
      alt="Everything Is AWESOME" 
      style="width:400;">
      </a>
    </div>

Best,

sgtmahoy, 2022-05-03:
Thanks for the quick response,

I'm a Robotics student at UWE Bristol in the UK and have been spreading word to students and professors about your project as an alternative to the E-pucks we have.

When running via roslaunch i get this error: [cid:e02114e4-6494-4da5-9e59-41aa87f3e4c8]

When running using rosrun i get this, however nothing happens:
[cid:5fbe1998-2c43-4d03-87bb-b09eb6c9264d]

chmod gives no feedback:
[cid:1323671c-e0b2-4360-96c9-4e378a8ead2d]

Thanks,

[Everything Is AWESOME] <https://www.youtube.com/watch?v=IO7FPEwkZkE>

Best,

[Everything Is AWESOME] <https://www.youtube.com/watch?v=IO7FPEwkZkE>

Best,

sgtmahoy, 2022-05-03:
Apologies! I spotted my own error, It was the urdfdom-py that was missing!

rezeck, 2022-05-03:
Hey @sgtmahoy,

This is really cool! Thanks for sharing the project, and I hope it will be useful for you there!

Oh glad it worked!
If you find any other problems feel free to contact me!

I will update the tutorial in the next few weeks.

Best,

closed 2022-05-03 by rezeck