verlab/hero_common #12: Is MT3608 required to assemble Hero2.0?

Suptree, 2022-08-23:
Thank you for developing Hero2.0.

I am planning to use it in the future for my research on swarm robotics.

Currently, I am assembling Hero2.0 based on the parts listed in assembly_materials.pdf.

And I am trying to join the PCB Board (v2.6) and MT3608.

However, the sizes do not match.

I looked at cad(https://a360.co/36zKPdi) and found that MT3608 is not used.

Is MT3608 listed in assembly_materials.pdf not used?

Also, what should we use as a replacement component?

I am not an expert in the mechanical field, so please be gentle and let me know.

best regards, and thank you in advance for your help.

[image]
[image]
[image]

rezeck, 2022-08-31:
Hi @Suptree 

Thank you for considering this robot in your projects!

We recently switched from [another module](https://www.sparkfun.com/products/retired/8290) to this one, so the hero board is not fully compatible (sorry for this issue). I will update the board schematic soon! 

However, I recommend using the MT3608 since it can provide (theoretically) 2A, which is more than enough for the robot allowing power other modules (e-hats).

In this step, try to adjust the MT3608's potentiometer (and fix it with some glue) so that the output voltage is at 4.90V. Some WS2812 LEDs do not work properly at voltage > 5.0V.

To install it on your board version, we solder a cable (>= 24 AWG) as a connector between the board and the module. The module should fit below the nodemcu board and the hero board (as in the figure below). 

Any questions, feel free to contact me.

[image]

[image]

[image]

[image]

rezeck, 2022-08-31:
Hi @Suptree

I just noticed that some of the IR sensors are soldered upside down. As a guide, try to align the top view of the IR sensor with the drawing on the board. That chamfer should line up with the one drawn on the board.

[image]

Best regards

Suptree, 2022-08-31:
Hi @rezeck

Thank you for the clear explanation.

Okay, I will use AWG24 cable to bind the board to the MT3608 until the latest version of Hero borad is released.

Thanks for pointing out the IR sensor. I was not aware of it.

We will contact you if we have any further questions.

wt160, 2022-10-02:
Hi, @rezeck:
Great work on the robot!
Just to say hi and show you guys there are people out here trying to build the robot.
I met the same issue, will try the solution mentioned above, but still hope you may update the  board schematic.
Thanks in advance! 
[image]

rezeck, 2022-10-09:
Hi @wt160 and @Suptree,

Sorry for my delay. Unfortunately, it's hard to find time to focus on improving HeRo. But whenever possible, I try to fix or improve some points.

I just committed a new version (v2.7) of the PCB with support for MT3608, and I also positioned the battery charging status LEDs to improve the visualization. More information [here](https://verlab.github.io/hero_common/pcb_order/).

Unfortunately, I didn't have time to test this new version properly, but I believe everything is working fine. If you find some issues, please let me know.

@wt160 I liked the PCB with the purple mask! It's very cool!

Thank you so much for considering this robot for your projects! =)

wt160, 2023-01-02:
Hi, Rezeck:
I have used the v2.7 design, the board is like this.
[image]
Could you help me to check whether the TCRT5000 alignment is correct?
You posted a picture above, but I am not sure whether the green one is correct, or the red one is correct.

Thanks!
Wei

rezeck, 2023-01-05:
Hi @wt160,

Sorry for the late reply! I hope I've been able to keep up with the development.

It seems to me that you have to invert all the TCRT5000, as they are upside down. 

The photodiode (light blue led) must be aligned with that chamfer drawn on the board. The TCRT5000's plastic case, when viewed from above, has a chamfer on the side of the photodiode. So the idea is to align the chamfer on the case with the one in the board. Perhaps this figure will help you visualize.

[image]

Another tip, check if the TCRT5000 pull-down resistor is connected to the phototransistor (dark led).

[image]

Please let me know if it works for you.
Regards,
Rezeck

wt160, 2023-01-07:
Hi, Rezeck:

Thank you for your reply! The pictures are really helpful. I will try the correct way.

Best,
wt160