verlab/hero_common #14: Compile-time Error: "i2s.h: No such file or directory"

abrahamjsimpson, 2022-10-28:
I am working on a team that is using HeRo for a research project. We followed your steps [here](https://verlab.github.io/hero_common/firmware/) for installing and setting up Arduino and flashing the firmware to the onboard ESP8266. However, when we went to Verify/Compile it in the Arduino IDE we encountered the following error (note that I am abbreviating the full filepath with ". . ." for brevity):
`In file included from C:\Users\. . .\HeRo\hero_common\hero_firmware\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:67,
                 from C:\Users\Matthew\Documents\A_School\Capstone\Code\repos\HeRo\hero_common\hero_firmware\firmware\LEDStatus.h:34,
                 from C:\Users\. . .\HeRo\hero_common\hero_firmware\firmware\firmware.ino:37:
C:\Users\. . .\HeRo\hero_common\hero_firmware\libraries\NeoPixelBus_by_Makuna\src/internal/NeoEsp8266DmaMethod.h:42:10: fatal error: i2s.h: No such file or directory
   42 | #include "i2s.h"
      |          ^~~~~~~
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
` 
The problem seems to be that this file is looking for a header file called "i2s.h", which is not present in the project. However, there _is_ in the same directory a "Esp32_i2s.h" and a corresponding .cpp file. Changing line 42 in NeoEsp8266DmaMethod.h to instead import this file allows it to compile, but we are worried about this causing deeper problems later since HeRo uses the ESP8266, **not** the ESP32 (and indeed, although the code compiles, when uploaded the ESP8266 it does not work as intended, which may become the subject of another Issue on this page if this does not end up being the problem).
In the latest commit of Makuna's NeoPixelBus, NeoEsp8266DmaMethod.h does not import either of these files (or any of the 11 headers that the version in HeRo's repo does). Do you know that the specific version of NeoPixelBus that is currently in the HeRo repo is needed, or should we be using the latest version?

rezeck, 2023-01-07:
Hi @abrahamjsimpson,

Sorry for the late reply! I hope I've been able to find a solution.

After a while, I was able to replicate this error. 
I was using version 2.5.0 of the esp8266 board, and this error did not come out. 
[image]

By upgrading the esp8266 board to version 3.1.0, I was able to replicate the error. 
[image]

The solution was to update the NeoPixelBus library as you suggested simply. I was using NeoPixelBus version 2.5.0, which is compatible with esp8266 board 2.5.0. 

I've updated the hero repository with the latest version of NeoPixelBus, version 2.7.1, which works with the esp8266 board version 3.1.0.

Please let me know if it now works for you.
Best,
Rezeck

abrahamjsimpson, 2023-01-24:
Hi @rezeck 
sorry for _my_ late reply. It does now compile using version 3.1.0 and works! Thank you for your help!

closed 2023-01-24 by abrahamjsimpson