verlab/hero_common #16: Materials: Bultin Battery Charger Resistors

hauff, 2022-12-06:
Hello, I have a question regarding the assembly materials [hero_resources/assembly_materials.pdf](https://github.com/verlab/hero_common/blob/7be0daf3eff30f7db28b033778bef5dd4830c5d5/hero_resources/assembly_materials.pdf).

There are 3 resistors (`R12`, `R13`, `R14`) listed in the section `Bultin Battery Charger`.
Are the descriptions (1K Ohm, 100 Ohm, 120 Ohm) correct?
I am asking because I thought the code 103 for example means 10 kOhm? (and 102 := 1 kOhm, 122 := 1.2 kOhm)

| Part Name | Device | Description |
|---|---|---|
| R12 | Resistor 103 | SMD Resistor 1K Ohm |
| R13 | Resistor 102 | SMD Resistor 100 Ohm |
| R14 | Resistor 122 | SMD Resistor 120 Ohm |

rezeck, 2023-01-05:
Hi, @hauff,

You are absolutely right! Thanks for pointing this one out.
I have already updated the website and document with this information.
The correct is:
Part Name | Device | Description
-- | -- | --
R12 | Resistor 102 | SMD Resistor 1K Ohm
R13 | Resistor 101 | SMD Resistor 100 Ohm
R14 | Resistor 121 | SMD Resistor 120 Ohm

I have also checked all the other descriptions, and they seem correct. If you find another issue, please let me know.
Regards,
Rezeck

hauff, 2023-01-05:
Hi @rezeck,

Thank you for answering my question. 🙂

closed 2023-01-05 by hauff