verlab/hero_common #17: Reading the IR Sensor Data

jacobh2299, 2023-01-25:
Hello! I am working with a research lab that has manufactured 5 of your Hero 2.6 robots. We seem to have run into a few issues when it comes to running the example code that you have provided in your github. While trying to run the random.launch python script, our robot only rotates clock-wise. I have a few screenshots of the rostopic laser that was provided of our IRs while uncovered and while covered. Our understanding is that the IRs should be defaulted to 0.5, but it appears that they are not defaulting to that very well. The first image is of the IRs uncovered, while the second image is of the IRs covered. 
If you could help us understand what we are looking at in the laser rostopic or help explain the random.launch script that you had written I think it would help our debugging process run a bit quicker:

Uncovered IRs:
[image]

Covered IRs:
[image]

Note that we changed the python script for random.launch to use python3 as we were getting errors regarding Yaml while only using Python 2.x. Changing to python3 resolved the crash issue, but the bot still only rotates clock-wise while running the script. 

Thanks!
University of Utah Research Team

rezeck, 2023-02-13:
Hey @jacobh2299,

Sorry again for the late reply.

It seems to me that the laser is not working properly. In the second figure,  the ranges field should be a value close to range_min. The ranges field contains the distance (in meters) for 16 IR sensors. We interpolate these 16 IR sensors (blue) from the 8 real IR sensors (red) on the robot. The field intensities are the light intensity (0-1023) received by the photodiode minus the ambient light intensity. That is, it would be the intensity of light returned after reflecting on an object. The correspondence IR sensor and the position in the robot are given in the following.
[image]

This is a video of expected behavior.

https://user-images.githubusercontent.com/14208261/218348064-9638f3d9-a66f-43f9-814f-426c5cd91aa5.mp4

*In your case, it may be a problem due to a lack of calibration or hardware failure. Please, test the two procedures below and let me know if they worked for you.*

**Calibration**
To calibrate the 8 IR sensors, use the launch file ```hero_confmode.launch```, and the ros service ```/hero_*/ir_calibration```. 
The process of calibration is done as follows:

1) place an object (preferably white color) at a known distance (d) from one of the IR sensors you want to calibrate;
2) Use the laser topic ```rostopic echo \hero_*\laser``` to see the intensity for that sensor;
3) The calibration parameter is obtained by the equation (intensity) * d^2.
4) Repeat this procedure for all sensors and save these 8 values;
5) With the 8 calibration values in hand, run the calibration service:
```sh
rosservice call /hero_*/ir_calibration "{IR0: 0.39, IR1: 0.28, IR2: 0.45, IR3: 0.29, IR4: 0.52, IR5: 0.22, IR6: 0.23, IR7: 0.28}"
```

I am still improving this process and the following is an alternative. 
You must place 8 objects at the same distance from the 8 IR sensors. 
I used the #D printer to make [support](https://github.com/verlab/hero_common/tree/noetic-devel/hero_resources/3d_parts/ir_calibration) to help with this setup. 
[image]

After placing the objects, run the script:
```sh
rosrun hero_examples hero_ir_calib.py hero_*
```
This is a video showing how to execute this script.

https://user-images.githubusercontent.com/14208261/218351257-2bc2767a-8aff-4ed9-b512-e373bce10820.mp4

**IR Sensor Test**
If you still don't have any luck, proceed with this process to see if everything is ok with the hardware. Try burning the RangeSensor firmware, and check the output for consistency. Maybe it has an IR sensor soldered upside down.
This is a video showing how to proceed with this test.

https://user-images.githubusercontent.com/14208261/218352369-a99e7c69-c070-4dcd-ba82-57c5e348644f.mp4

Please let me know if everything works out.
I'm still trying to refine and automate this sensor and actuator calibration step. 
As soon as I finish this, I will update the tutorial.

Best,
Rezeck

jacobh2299, 2023-02-17:
Thank you for such a detailed reply! I found your commits in the noetic-devel branch and downloaded the new hero_ir_calib.py file. It seems to be running properly on our end, but our values displayed from rostopic laser still appear to not be calibrated (likely because we do not have well measured distances when calibrating). I am currently 3D printing the part you made to assist in calibration and will update this post when completed. 

Thanks again
Jacob, University of Utah Drew Research Lab (DRL) Swarm Robotics

rezeck, 2023-02-22:
Hey @jacobh2299 ,

Yes, the IR positions between version 2.5 and >= 2.6 are different. This is because there is a design error in board version 2.5. To (temporally) fix it, I have to flip the TCRT5000 and also the IR and phototransistor inside the plastic case. Anyway, I definitely suggest NOT using version 2.5. I will put this warning in the tutorial.

Since you are using board version 2.6, this problem has been fixed. So I think the way you placed the TCRT5000 (as shown in your figure) is correct.

When soldering the TCRT flipped in version 2.6, it may probably burn the IR. You can check the IR using the cellphone camera (with no IR filter) or a webcam. It should blink a purple light. To test the phototransistor, try putting the multimeter (DC voltage 0-5V) as in the figure. 
[image]

Cover and uncover the phototransistor. If the voltage consistently changes, it should work fine. Anyway, if you have another TCRT5000, try to replace one of them and test it.

Let me know if you find something or if you have more questions.

Best,
Rezeck

closed 2024-02-15 by rezeck

Albulinek, 2024-10-01:
@rezeck 

The testing firmware on 2.7 showcased in your video is from branch "melodic-devel"

I am not sure if there was some mixup of files for "master" and "noetic-devel", but testing RangeSensor firmware in those branches does not produce any readable values from python script.

I was searching topics here after failing to bring up Hero to WiFi hotspot mode. I will also check original firmware itself. I had to comment out config check part and hardcode Hero bringup to configure it properly.

Lastly, the material assemby was missing some crucial parts, along with wiki is missing details about PCB assembly. I am not sure if this is intentional, but once I am done polishing Hero on my side, I will try to open PR to help others with building their own bot.

Thanks for providing us with this awesome resource!