verlab/hero_common #18: Rosservice List not displaying expected results

jacobh2299, 2023-02-01:
Hello again! Another question for you. We are not getting an expected rosservice list as shown in your instructions website. We expect to see several services that you have included such as pid_calibration, set_frequency, set_odom, and set_pid_parameters, but we are only receiving 
/rosout/get_loggers
/rosout/set_logger_level
/rosserial_server/get_loggers
/rosserial_server/set_logger_level

Here is a screenshot of our computer connected to a robot, the rostopic list, and the rosservice list:

[image]

Here is also a screenshot of our catkin_make return terminal:
 
[image]

We have tried re-flashing the firmware on the robots with the ESP version 2.5 such as you recommended before, but with the current version of firmware, using the latest version of the board has been working. We also tried re-cloning the repo to a different computer running Ubuntu and the service list was still not showing. I would also like to note that we did not use the Docker Method but rather built from source, although I don't think that this is a problem.

Best,
University of Utah Robotics Team

rezeck, 2023-02-09:
Hey @jacobh2299,

Sorry for the late reply.

Lately, it has been hard to commit myself to develop the robot's tutorial, but whenever I get some time, I complete and improve it.

About the issue above, the problem is with the rosserial package. The official version has two implementations: one in Python and another in C++.

The C++ version implements communication only via topics but supports multiple robots. The Python version implements communication via topics and services but does not support multiple robots.

A while ago, I was trying to debug and add communication via services in the C++ version of rosserial. You can find some modification that may worked for a while, but mostly I find stopped working ([Issue](https://github.com/ros-drivers/rosserial/pull/453)).

As an option, I decide to leave the robot services only for configuration. Since then, two launch files have been able to bring up the robots: ```hero_bringup.launch``` and ```hero_confmode.launch```.

The first runs the C++ version of rosserial and works for multiple robots. The second runs the Python version and enables communication via the service.

So, I assume you ran ```hero_bringup.launch``` instead of ```hero_confmode.launch```. Can you confirm this?

Best,
Rezeck

[image]

jacobh2299, 2023-02-10:
I can confirm this! I was using the hero_bringup.launch, thank you for the explanation! We have sense used the hero_confmode.launch, and with some light edits to the common.xacro file we were able to get all the services running. I will link our edits to common.xacro here.

The main reason we are looking into the services was because you have a service listed as IR_Calibration, and we seem to be experiencing many issues regarding the IRs. Hopefully with more experimentation on the values we receive from the IRs we will be able to run the random.launch file to run properly. If you could tell us what the IR_calibration service does and how it should effect the values that we receive from the IRs, that would really help streamline our debugging process. Thanks!

Edits to common.xacro: 

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <!-- Some properties of HERO -->
  <xacro:property name="PI" value="3.1415926535897931" />
  <xacro:property name="wheel_separation" value="0.06460" />
  <!-- <property name="wheel_diameter" value="0.0496" /> -->
  <xacro:property name="wheel_diameter" value="0.050" />

  <xacro:macro name="cylinder_inertia" params="m r h">
    <inertia ixx="${m*(3*r*r+h*h)/12}" ixy = "0" ixz = "0" iyy="${m*(3*r*r+h*h)/12}" iyz = "0" izz="${m*r*r/2}" />
  </xacro:macro>

  <xacro:macro name="box_inertia" params="m x y z">
    <inertia ixx="${m*(y*y+z*z)/12}" ixy = "0" ixz = "0" iyy="${m*(x*x+z*z)/12}" iyz = "0" izz="${m*(x*x+z*z)/12}" />
  </xacro:macro>

</robot>

```

rezeck, 2023-02-13:
Hey @jacobh2299 

Excellent!

Sure, I just replied to your issue #17. 
Let me know if this may help you guys with the IR sensors problem. 
Then check if the random walk algorithm works.

Thanks for reporting this issue.

Best,
Rezeck

closed 2023-02-17 by jacobh2299