verlab/hero_common #20: TF issue in the launch file generated from gazebo_wizard

Suptree, 2023-09-11:
Hello, I've been conducting research using Hero2.0. It's an invaluable tool, and I'm very grateful for it.

## Issue Overview:
[image]

I have concerns regarding the TF description in the launch file generated based on the wizard screen initiated by the ```roslaunch hero_gazebo gazebo_wizard.launch``` command. While the Gazebo simulator operates without any problems, an error arises when attempting to display in rviz.

[image]

## What I Checked:
To investigate the TF relationships, I executed the following command:
```$ rosrun tf2_tools view_frames.py```
From this, I was able to observe the following TF relationships:

[not_fixed_tf.pdf](https://github.com/verlab/hero_common/files/12571253/not_fixed_tf.pdf)

Is this the intended behavior?

## My Modification and Proposal:
I made changes to wizard_script.py and corrected the launch file. As a result, the TF relationships were as follows:

[fixed_tf.pdf](https://github.com/verlab/hero_common/files/12571256/fixed_tf.pdf)

With this modification, HeRo was successfully displayed in rviz.
[image]

If the aforementioned TF relationships weren't intended, I would like to submit my modification as a pull request.

Thank you for your consideration.

rezeck, 2023-09-12:
Hello @Suptree ,

Thank you for reaching out. I am thrilled to hear that you find it useful for your research.

Regarding the concerns you've raised about the TF description in the launch file generated by the wizard screen, I appreciate your thorough investigation and the detailed information you've provided. 

The corrected TF relationships are the ones shared in "fixed_tf.pdf". This error may be due to rule changes in tf-prefix between ros noetic and melodic distros. As I have been using the physical platform more, I ended up not seeing this in the simulation.

Anyway, your willingness to contribute by submitting a pull request is greatly welcomed. 

Once again, I truly appreciate your feedback, and if you have any further questions or suggestions, please don't hesitate to let me know.

Thank you for contributing to this project!

Regards,
Rezeck

Suptree, 2023-09-13:
Thank you for your response, @Rezeck.

I appreciate you identifying the issue related to different ROS distros.
I've branched off from the `noetic-devel` branch to make the necessary changes.
I've submitted a pull request for your review.
Thank you for your consideration.

Regards,
Suptree

closed 2024-02-15 by rezeck