verlab/hero_common #22: Short circuits (Vin/VCC and GND; MUX_X7 and GND) in latest PCB design (v2.7 2023-03-05)

Honza0297, 2023-10-08:
In the latest version of the PCB (v2.7, 2023-03-05), there is a short circuit between VCC (specifically VCC pin of capacitor C1) and GND (specifically GND pins of 4051) via B+ route (highlighted in the image). In fact, the whole B+ route seems misplaced, crossing several other routes in the same layer. Please correct me if I am wrong.

[image]
 
The issue is present in the real PCB as well. Actually, another issue is present in the physical PCB: route MUX_X7 is shorted to GND (GND pin of capacitor C1). The position itself of of the C1 capacitor seems unfortunate, colliding with Neopixel LED as well.

rezeck, 2023-10-08:
Hi @Honza0297 

Thank you for bringing this critical issue to our attention. I sincerely apologize for any inconvenience this may have caused. It appears there may have been an error during the upload of updates for version 2.7. This layout is not supposed to be of the PCB version 2.7.

# Version 2.7 (2023.10)
[image]

Gerber: [board_2023-10-08.zip](https://github.com/verlab/hero_common/files/12841293/board_2023-10-08.zip)

I have now uploaded the corrected files for version 2.7, and after careful review, I did not identify any further issues. 
Your feedback is invaluable, and if you have any additional suggestions or improvements for the PCB layout, please feel free to submit a pull request. We greatly appreciate your contribution.

Once again, I apologize for any problem this may have caused, and I appreciate your understanding.

Best regards,
Rezeck

Honza0297, 2023-10-08:
Wow, thank you for the fast fix, @rezeck! I sent the fixed version to production and once the new PCBs arrive, I'll let you know whether everything works.

Honza0297, 2023-11-15:
I finally managed to test everything and PCB seems working flawlessly. :) 

However, I have run into another issue. When I upload the firmware and try to teleoperate the robot with:

```
roslaunch hero_bringup hero_bringup.launch
roslaunch hero_bringup hero_teleop.launch id:=0
```   

the robot behaves pretty chaotically, both in real life and rviz. TFs are moving according to real movement etc., but when I command it to go straight, it just chaotically moves around. Am I supposed to tune some parameters? Thanks in advance!

rezeck, 2023-11-16:
Hi @Honza0297 ,

Thank you for your reply. It's great that PCB is working now.

Yes! I do not know if you pass through the calibration step [here](https://verlab.github.io/hero_common/calibration/). If not, try calibrating the motor's PWM endzone and the PID parameters for both the position and velocity controller. You should set the PWM endzone with the values that fully stop your motors (theoretically 1500). The PID params you can use the ones in the tutorial, as they should be reasonable to start using. 
> PS.: Use this bringup ```roslaunch hero_bringup hero_confmode.launch``` instead.

Also, when you are prompted to command the robot using hero_teleop, try to decrease the linear speed to ~0.1 m/s. Maximum is 0.20 m/s.

If the robot is turning around when you command it forward (keyboard ```I```), you may check these points:
- Motor calibration and PWM parameters?
- Is the encoder speed filter working properly (Tip: use rqt to plot them)?

Please let me know if you manage it to work.

> PS: To calibrate the IR sensors, follow this [issue](https://github.com/verlab/hero_common/issues/17). I could not find time to add this information to [webpage tutorial](https://verlab.github.io/hero_common/).

closed 2024-02-15 by rezeck