verlab/hero_common #25: can i replace CD4051BM

buddytex, 2024-02-10:
can i replace CD4051BM without another mux, if so how to determine please help

rezeck, 2024-02-15:
Hi @buddytex 

The CD4051BM can be operated on a wide range of supply voltages, anywhere between 3V and 20V (we are using 5V supply). Another alternative is 74HC4051, which only operates from a 5V supply. To determine if it is compatible, you just have to check the package (SOIC-16), supply voltage (5V), and the pinout to see if it matches with the CD4051BM. Otherwise, you have to modify the PCB routing traces.

buddytex, 2024-02-24:
I dont know where to the place charging module and step up module, I cant find the resources or images to place it

rezeck, 2024-03-02:
Hi @buddytex ,

Apologies for the delay in responding. I've been juggling multiple tasks lately, making it challenging to allocate time to complete the HeRo tutorial.

Regarding your query about the step-up module, I recommend referring to the details provided in issue #12. Hopefully, it sheds some light on the placement of the step-up module onto the board.

As for the charging module (TP4056), it's soldered directly onto the board. I've attached an image below illustrating where the components should be placed for your reference. 

[image]

Feel free to reach out if you need further help or clarification on any aspect of the project.

Best regards,
Rezeck

closed 2024-04-01 by rezeck