verlab/hero_common #33: gazebo algorithm issue

besobees, 2024-03-15:
Hello @rezeck,

We've tested your impressive GRF-based collective transport algorithm, but unfortunately, it seems to be ineffective. When the robots detect the target, they all converge on it regardless of the presence of objects. Could you please send us the updated files?

Additionally, we've constructed your Hero 2.0v robot and would like to implement the algorithm on this model.

Looking forward to your response.

rezeck, 2024-03-20:
Hi @besobees 

It's fantastic to hear that you've constructed the Hero robot! We're always thrilled to see our work coming to life. If you have any images or feedback to share, please don't hesitate to do so. We greatly value input from our community of contributors.

As for the issue with the GRF-based collective transport algorithm, I'd recommend revisiting the initial issue description on our GitHub repository at https://github.com/rezeck/grf_transport/issues/1. If the issue persists and it's related to the code within this repository, we can continue troubleshooting it here.

Looking forward to resolving this matter with you.

Best regards,
Rezeck

besobees, 2024-03-27:
i will send you my problem here https://github.com/rezeck/grf_transport/issues/1 take a look @rezeck <3

rezeck, 2024-04-01:
Solved #33 in https://github.com/rezeck/grf_transport/issues/1

closed 2024-04-01 by rezeck