verlab/hero_common #34: PID Controller Issue

Khaled3ly, 2024-04-21:
Hello @rezeck,

I can't understand how does the PID controller work as the input signal is velocity (cm/s) and encoder reading is also velocity (cm/s) while the output is PWM.... is there a transfer function that takes output (cm/s) and transfers it to PWM ?

Thanks in advance.

[image]

varunraveendra, 2024-06-30:
Hi @Khaled3ly ,
I have been working on Hero robots for a while now. Talking about the PID controller, I believe that it takes in the desired and the current velocity and outputs adjusted velocity. This is then summed with the MotorDeadZone value. If you are looking at raw values of the adjusted velocity, then you might notice that it can go up to +250 or -250. The code below  under VelocityControl.ino might give you a better explanation:
[image]

I hope this helps.

-Varun Raveendra

rezeck, 2025-09-27:
Hi @Khaled3ly, we control PWM by closing a velocity loop: the encoder measures wheel speed, the PID compares it to the target speed, and its output is treated as the actuator effort command that the driver maps directly to PWM duty. Here the PID already returns a bounded value in the driver’s units, we add a small dead-zone offset to overcome static friction, and send that to the motor driver—no extra transfer block is needed, just tuning (PI first, with saturation/anti-windup). If this clarifies it, I can close the issue.

thanks @varunraveendra